Open-STEM/XRP_MicroPython #1: Stopping the motors on program exit

bradamiller, 2023-06-07:
It would be nice if there was a way to stop the program on exiting. This would mean, setting all the motor speeds to 0. That implies that the program is running in some framework that cleans up at the end of execution, or the user is required to add a stop_all_motors() call at the end of the program. The framework would be better so that users don't have to remember.

Kreeevin, 2023-06-15:
I agree that we should look into this for the future. For now, the users can just include a `drivetrain.stop()` call to the end of their program.

bradamiller, 2023-07-04:
I like that as long as the IDE stop button can stop the motors from running. I think it does that but frank would have to confirm that.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

fgrossman, 2023-07-04:
> Sent with GitHawk <http://githawk.com>

I think it is good habit to stop your motors when done. But agree with Brad
that the stop button is important. I think having a resetbot.py is fine for
now. We can always improve in the future.

   Frank

closed 2023-07-09 by Kreeevin