Open-STEM/XRP_MicroPython #2: Determine if the "on" switch is not on

fgrossman, 2023-06-14:
If you run a program from the IDE or when the robot is connected to the computer, you may have power to the Pico W, but no power to the motors. If a program is run in this state, there is a good chance that the libraries will just hang waiting for encoders and the IMU. We should recognize that situation and print something out to help the user solve the problem.

Kreeevin, 2023-06-15:
I thought the board already had an indicator light for if there was power to the motors, separate from the one that indicated if the board itself has power? I'm not sure if there's a super clean software implementation for this, and from my understanding the hardware solution is already in place

fgrossman, 2023-06-17:
The light is there, but I constantly observed people being frustrated. The two LEDS are right next to each other and people do not check them. They just get frustrated that their program is not working. 

We can tell if the switch is on with the ADC on gpio 28

vin = machine.ADC(28)
vin.read_u16()

This is the voltage from the batteries to the motors.

Kreeevin, 2023-06-17:
Just pushed board.py, which has that functionality. after importing defaults the user can now run `board.are_motors_powered()` to get a bool.