Open-STEM/XRP_MicroPython #3: Time out if library can't get to its condition

fgrossman, 2023-06-14:
If the motor values are too low, or something is blocking the robot, or the robot is up on blocks we may never meet conditions that are being looked for. Usually encoders or IMU. We should add timeouts for these types of calls.

Kreeevin, 2023-06-15:
The basic drive methods (straight, turn) already have timeout functionality as an optional parameter. I think it's reasonable to leave the use of that parameter up to the user?

fgrossman, 2023-06-16:
I could argue the other way around. Since this platform will be used by
novices, we don't want them to get too discouraged when their program never
finishes. I guess if we document putting in the time out value in the
lessons and we always include it in the code generated from the blocks.
Brad, are you planning on including it and explaining the timeout in the
curriculum?

Kreeevin, 2023-06-17:
@bradamiller

bradamiller, 2023-07-04:
My thought is that there are already timeouts on motor commands. But often you want timeouts on operations completing, like if a ore complex task has completed. I would suggest that we add a class like timer (that name is probably taken) that has two methods: timer.set_timeout(time-in-ms) and timer.is_timed_out(). You call the first one before the loop, and use the second one in the exit condition for the loop. Then the loop could exit on whatever the condition is OR if the timeout occurred.

For operations like turn(angle) where the loop is in the library, we would make sure that there are optional timeout argument like on the drive methods as Kevin said.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Kreeevin, 2023-07-08:
Timeout class has been added and implemented inside of the DifferentialDrive class. I plan to add some example code of how to use it soon (it's pretty intuitive regardless), but for now I'm marking this issue as closed.

closed 2023-07-08 by Kreeevin