Open-STEM/XRP_MicroPython #18: Random Ultrasonic Spikes

milespearlman, 2023-07-07:
When the robot was about 30cm away from an object and the ultrasonic was overall reading accurately, the sensor will randomly spike for about 3 cycles, usually + or - about 20cm but sometimes even more. 

[image]

[image]

milespearlman, 2023-07-07:
Note that this occurred only when the motors were on.

Kreeevin, 2023-07-08:
Wait this only happens when the motors are on? can you try to reproduce this bug now that I've fixed some of the issues associated with the encoders?

Kreeevin, 2023-07-08:
Ok so I've reverted an older change to the rangefinder code to remove the source of this error. The trade off is that the rangefinder code is now back to being blocking, which means that it interrupts the flow of the program for potentially up to 30ms (which should basically never be that extreme). This isn't a permanent solution but it does make the code functional until the other implementation has been fixed.

closed 2023-07-08 by Kreeevin