Open-STEM/XRP_MicroPython #19: Extreme Ultrasonic Distance Spikes

milespearlman, 2023-07-07:
Occasional very extreme spikes when working with proportional control that would cause xrp to go significantly too fast (code below)

[image]

Kreeevin, 2023-07-08:
This is reporting the same error as Issue #18, correct? If so, I am already looking into it and we can close this issue for now

Kreeevin, 2023-07-08:
Copied from the comment on #18:
I've reverted an older change to the rangefinder code to remove the source of this error. The trade off is that the rangefinder code is now back to being blocking, which means that it interrupts the flow of the program for potentially up to 30ms (which should basically never be that extreme). This isn't a permanent solution but it does make the code functional until the other implementation has been fixed.

closed 2023-07-08 by Kreeevin