Open-STEM/XRP_MicroPython #27: Should the index be consistant between encoded_motor and encoder?

fgrossman, 2023-07-12:
It is odd that the encoders are 0 based index and the encoded_motors are 1 based index

Kreeevin, 2023-07-12:
The reason for this is because I wanted the motor indexes to line up with the labels, so index 3 would correspond to motor 3. I can change this fairly easily so lmk if you think this shouldn't be this way

fgrossman, 2023-07-12:
I was not saying it was bad to be 1 based to line up. I was saying that
encoder should be the same way.

Kreeevin, 2023-07-12:
The index used for the encoders is for which PIO state machine we are using, of which were are 4, indexed 0 through 3. We can't really change that to be one-based indexed because that's straight from the board. What I will do is add more documentation to the encoder class so that the meaning of the index is more apparent.