Open-STEM/XRP_MicroPython #30: Reading the rangefinder back to back gives bad results

fgrossman, 2023-07-12:
The following code, the first read gets a good value, the second one gets 65535. If I add a sleep on both sides of the print then both reads get valid information.

`from XRPLib.rangefinder import Rangefinder
import time

rangefinder = Rangefinder.get_default_rangefinder()

while (rangefinder.distance()) > 10:
  print(rangefinder.distance())
  time.sleep(1)`

Kreeevin, 2023-07-12:
I attempted to replicate this bug but was unable to do so. Is there anything else you can share about your setup for this bug?

Kreeevin, 2023-07-12:
[image]
[image]
Here's my printout from trying to replicate your bug; I can try to replicate this again but I'm really just not seeing the same issue you are seeing