Open-STEM/XRP_MicroPython #31: Encoder response seems very slow

fgrossman, 2023-07-12:
With the following code below, the numbers would print 3 more times at the same value after I turned a wheel, and then print out the correct value.

```
from XRPLib.differential_drive import DifferentialDrive
import time

differentialDrive = DifferentialDrive.get_default_differential_drive()

while True:
  print('Left ' + str(differentialDrive.get_left_encoder_position()))
  print('Right ' + str(differentialDrive.get_right_encoder_position()))
  if (differentialDrive.get_left_encoder_position()) > 20:
    differentialDrive.reset_encoder_position()
  time.sleep(1)
```

Kreeevin, 2023-07-12:
From the brief testing I have done, I'm seeing similar issues with the slight delay but not to the same extreme. I tried testing with various printing rates and I was seeing that 3-print delay no matter what the print rate is. Any chance it's some issue with the serial buffer instead of an actual delay with the encoder?

fgrossman, 2023-07-12:
I don't think so. I don't see a delay at those speeds with the rangefinder or button press.

Maybe try something with a timer and a button press until it sees a change in encoder values.

Kreeevin, 2023-07-14:
Just pushed a kind of sketchy change that fixes this and the other two associated issues. I wish there was a better solution than just reading it 5 times but this definitely works, so I'm going to mark this issue as closed for now

closed 2023-07-14 by Kreeevin