Open-STEM/XRP_MicroPython #36: Endoded_motor reset_encoder_position does not seem to be working

fgrossman, 2023-07-14:
The code below the number of ticks stays the same for both printouts

```
from XRPLib.encoded_motor import EncodedMotor
import time

motor1 = EncodedMotor.get_default_encoded_motor(1)

motor1.set_speed(60)
time.sleep(2)
motor1.set_speed()
print(motor1.get_position_ticks())
motor1.reset_encoder_position()
print(motor1.get_position_ticks())
```

Kreeevin, 2023-07-14:
Same issue as #37 and therefore #31. There's a delay in when the update actually goes through:
[image]
[image]

Kreeevin, 2023-07-14:
see comment on #31, solved for now

closed 2023-07-14 by Kreeevin