Open-STEM/XRP_MicroPython #37: Encoded_motor ticks only work with set_speed

fgrossman, 2023-07-14:
The code below gives a tick count of zero.  If I use set_speed the tick count advances

```
from XRPLib.encoded_motor import EncodedMotor
import time

motor1 = EncodedMotor.get_default_encoded_motor(1)

motor1.set_effort(0.8)
time.sleep(2)
motor1.set_effort(0)
print(motor1.get_position_ticks())
```

Kreeevin, 2023-07-14:
[image]
I don't love this answer but I think this is caused by the same thing as #31. After 3-4 prints, it then actually updates to the next jump in values. 
[image]

Kreeevin, 2023-07-14:
[image]
extra screenshot to show minimal reproduction steps

Kreeevin, 2023-07-14:
see comment on #31, solved for now.

closed 2023-07-14 by Kreeevin